borb/amigahid-pico #13: Check all international keymap variations

borb, 2022-04-17:
The Amiga keyboard uses rows/columns to select keycodes, yet still use the same firmware on the 6507 MCU. This means the membrane matrix can add additional keys without needing firmware support. However, we don't have that just yet because a) I don't have a bunch of keyboards of different locales (only UK and US right now) and b) mapping to the Amiga "deadkeys" isn't entirely clear right now.

Commit c916c8e0d4e3d95ace3759338d88ed55e6f02100 added return-cutout (that's literally what it's called in the developer documentation), and mapped the UK HID `\|` to the Amiga `\|` (unfortunately the positioning was ideal for leftshift-cutout, again what it's called in the dev docs). So this means lshift deadkey is currently not mapped to anything.

I'm likely going to need help from Amiga users who have keyboard layouts other than UK & US.

rfuest, 2022-04-17:
I did some more tests and got most of the keys working by changing two HID to Amiga mappings. I'll report back with the exact changes I made once I'm back on my other development machine.

But even after fixing the mapping there were some remaining issues. Some keys on the German layout, like `[` and `]` are only accessible using the `Alt Gr` key. On the Amiga keyboard these symbols are located on the upper left corner of the number pad, but these keys aren't available on a PC keyboard. This would be hard to fix in the firmware and is also a problem which exists in emulators like WinUAE. One solution is to load a custom keymap on the Amiga, like the one shipped with WinUAE in the `Amiga Programs` directory.

rfuest, 2022-04-21:
I had to make these changes to `mapHidToAmiga`:
```c
mapHidToAmiga[0x31] = AMIGA_INTLRET;
mapHidToAmiga[0x64] = AMIGA_INTLSHIFT;
```
According to https://deskthority.net/wiki/Scancode 0x31 is always used instead of 0x32.