borb/amigahid-pico #7: USB Type A socket

borb, 2022-04-17:
Add a USB type A socket. This can be done by either:

* connecting to the micro-USB port and providing a type A socket; hacky but stable
* attaching to test points beneath the micro-USB socket and running to a type A socket

borb, 2022-04-17:
Added in revision 4 (see 65ffd97fab697f31d4ec6cc0d69d6c9d1e991e65). Awaiting PCB delivery to test.

borb, 2022-04-20:
Protos arrived today. Didn't specify castellated edges, but the finish is easily acceptable (the aperture is super tiny though, so this is either a hot air or tiny tip job. Had to trim some hanging via plating with a scalpel. Test pads line up ok, solder mask looks suitable. Think this might work ok.

Will assemble tomorrow.

borb, 2022-04-21:
Ok, it works.

[image]

I'm keeping it external whilst I probe around the pins to see what's firing where, but everything works. Still need to move the ground plane via, which is in the Pico's keepout area. But success.

rfuest, 2022-04-21:
Nice, the USB A connector looks so much better than an OTG adapter. I think I should build one.

borb, 2022-04-21:
> Nice, the USB A connector looks so much better than an OTG adapter. I think I should build one.

I have more boards than I am going to use in this production run. Would you like a couple of samples sent to you?

rfuest, 2022-04-21:
Thanks for the offer. I'm not sure where you are located, but unfortunately it's often cheaper to order new PCBs for China than paying postage for PCBs.

borb, 2022-04-21:
Socket is working. Closing.

closed 2022-04-21 by borb