borb/amigahid-pico #17: lcont2_h connected to RUN input

rfuest, 2022-04-21:
In the latest schematic revision the `lcont2_h` signal is connected to `RUN` input (pin 30) on the Pico. This would reset the Pico on mouse movement or when the joystick is pushed back.

borb, 2022-04-21:
🤦 

[image]

I really should have spotted that.

borb, 2022-04-21:
Okay, it's not as bad as resetting the Pico: the Amiga would have to pull back, and that won't happen because it's an output only and the Amiga keeps it at 5V; hence the other side of the fet will still be held at 3.3V.

So it's just a completely useless pin. Will fix; another revision on the way 😄

rfuest, 2022-04-21:
I was confused when I opened the issue and somehow thought that the controller connections were inputs :confused:.

rfuest, 2022-04-21:
Can we decide how this should be fixed in the next revision. I would like to change this in the PCBs I order and don't want to create yet another pinout variant.

The two easiest options are:
* Move `lcont2_h` to pin 32
* Move `lcont2_v` to pin 32 and `lcont2_h` to pin 31

borb, 2022-04-21:
The second option makes the most sense to me, since the first would mean having to route the second controller H line through another via in order to get to the same pin on the IDC connector, whereas the second just moves the fet connectors up a level.

Alternatively, I could produce another version this evening (I need to go and put my children to bed first) and if it passes your visual inspection, go with that version?

borb, 2022-04-21:
Actually, looking at the pcb layout, I'll just change the top-side tracks from the vias to the pads; the vias can remain where they are for bottom-side simplicity, and the top side can move over.

[image]

And I'll fix this as well. And remember not to turn the visibility of User.Drawings off whilst placing vias in future.

[image]

rfuest, 2022-04-21:
> Alternatively, I could produce another version this evening (I need to go and put my children to bed first) and if it passes your visual inspection, go with that version?

I think I'll create my own version of the PCB. I want to rearrange the keyboard and joystick connectors to one side of the PCB. And I'll probably try to use a 4 layer PCB to make the power/ground routing a bit easier. By using a 4 layer PCB it's also possible to use a controlled impedance trace for the USB connection, which is probably not necessary for the short USB 1.1 run, but it makes me feel better :wink:.

borb, 2022-04-21:
> I think I'll create my own version of the PCB. I want to rearrange the keyboard and joystick connectors to one side of the PCB. And I'll probably try to use a 4 layer PCB to make the power/ground [...]

You sound like you know what you're doing 😄 Would very much appreciate seeing this done properly as well!

rfuest, 2022-04-21:
Have you decided on a license for the PCB project yet? To my very limited legal/licensing knowledge, the non commercial clause on the PCB doesn't match with the EPL mentioned in the README.

closed 2022-04-21 by borb

borb, 2022-04-21:
> Have you decided on a license for the PCB project yet? To my very limited legal/licensing knowledge, the non commercial clause on the PCB doesn't match with the EPL mentioned in the README.

My knowledge is limited in this area also.

All I particularly want to avoid is the swathes of overpriced stuff being sold at a markup on eBay and elsewhere, especially since this is a fairly simple board. I've seen some other projects of similar ilk selling for ridiculous prices so I want to avoid that happening here.

My typical go-to would be creative commons, non-commercial, sharealike with attribution.

commit e354c984 (2022-04-21): revision 5 pcb (fixes #17 and #18)