borb/amigahid-pico #19: LED pins

rfuest, 2022-04-21:
The README states that the user must find some other ways to connect the pin 7 and pin 8 leds. IMO it would be useful to provide some option to use these signals. Here are some options to use them:
* Empty LED and resistor footprints on the PCB that can be populated if required
* A resistor on the board and two 2 pin 2.54mm pin headers to connect PC case style LEDs
* Connect the LED pins to the unused Pico inputs
  The LED state could be displayed on the OLED. The connection would require level shifting from 5V to 3.3V. If an ADC input is used for the Power LED it would even be possible to distinguish between the filter on and off states.

borb, 2022-04-22:
This is something I'm personally going to need to manage as the Checkmate case will need to display power state, and possibly drive state as well. Ultimately I'd like this to be a pass through for my own (and other people's) purposes, but feeding one or both into the Pico is certainly a possibility.

User @M1kerochip has suggested displaying the floppy status on, for example, the scroll lock LED, but I think I'd like to avoid that if possible - it has benefits for remote-operated Amigas but beyond that, I'm not an enormous fan.

Three of the four unused GPIO pins at the bottom are personally earmarked for rotary encoder emulation - it uses the same encoding method as a single ball-mouse axis so isn't a huge stretch to implement, and could relieve my complaints about using the microswitches on Gotek boards. However, if this was not to happen, then shifting the second controller pins down to the bottom pins to free up the ADC pins could be a possibility.

So, choices are:
- Into the Pico to display on the OLED
- Provide 3.3V/5V headers on the board for connecting to other LEDs

rfuest, 2022-04-22:
> Provide 3.3V/5V headers on the board for connecting to other LEDs

I'm not sure I understand what you mean by this. How would a 3.3V or 5V header help to connect LEDs?

borb, 2022-04-22:
> > Provide 3.3V/5V headers on the board for connecting to other LEDs
> 
> I'm not sure I understand what you mean by this. How would a 3.3V or 5V header help to connect LEDs?

The case has two sets of cables; one for Amiga-type systems, where the power source is 5V. The other is PC-type systems, where the source is 3.3V. This affects what voltage stepping is done on the front panel.

borb, 2022-04-22:
...I'm referring to putting standard 2.54mm pitch headers on to voltage shift or pass through the Amiga's LED signal.

rfuest, 2022-04-23:
Sorry, but I still don't get it. Do we agree that the Amiga 500 and the amigahid-pico will still be connected with 8 pin connection that includes the LED signals? And that the amighid-pico PCB would contain two 2.54mm pitch pin headers to connect PC case LEDs for power and disk activity? But in this case the voltage wouldn't matter, because PC case LEDs are raw LEDs which require an external current limiting resistor.

borb, 2022-04-24:
> Sorry, but I still don't get it. Do we agree that the Amiga 500 and the amigahid-pico will still be connected with 8 pin connection that includes the LED signals? And that the amighid-pico PCB would [...]

Yes, that's what I'm trying to say.