borb/amigahid-pico #23: A600 Pins

Frogowitsch, 2023-03-10:
Hello,
I had no problem attaching the device to my A500, but the A600 is not so easy.
SMD soldering is no problem for me, but before I destroy the A600, where exactly do I need to solder the wires to ? A picture would be great, but if not possible, please give me a hint. My board is a rev 1.5, if you need this information.

borb, 2023-03-10:
Hello! Unsure which board revision you're using but great to hear from a user.

So, the A600 differs from the A500 since the keyboard controller is on the main A600 PCB rather than the keyboard assembly, as per the A500, A1000, A2000 and other big-box Amigas.

Directly below the keyboard membrane connector and immediately to the right is U13, the keyboard controller for the A600. This is probably marked 6571or 68HC05 (the latter was used in the A1200 & A600, but the former pretty much only in the A600).

Here's a very crude annotated photo:
[image]

hid-pico is setup to use open drain/active low, so the two controllers should cohabit safely without you needing to remove power from the 6571/68HC05, though if I'm honest, with both controllers connected (hid-pico will send an init sequence to the Amiga) I have _no_ idea if the init sequences will collide with each other. If yes, you might want to consider finding a safe way to remove 5V from the 6571/68HC05, or alternatively I could put a sleep into the code to pause before sending init to the Amiga to prevent collision.

I would suggest getting hold of a 44-pin PLCC socket and placing it over the top, then tapping the connections from there.

Be sure to test the signal points (DAT, CLK, RST) for 5V before attaching. And check ground obviously.

borb, 2023-03-10:
For reference: Use https://amigapcb.org to confirm any or all signals before attaching. The rev 2A is extremely similar to the 1.5.

Frogowitsch, 2023-03-12:
I checked all pins on this website, seems fine. 
Now anything is connected, but it does not work. Where should I insert the sleep function ? Possible, that there is an init collision.
I checked the pins with my oscilloscope, dat, clk are fine, rst should also be.

Frogowitsch, 2023-03-12:
What I did not try is :
Can I only use the two joystick ports ? So I use the amiga keyboard and only convert mouse and joystick to usb. Is this possible ? Cant try atm, will be home in 4 days and would like to know :)

borb, 2023-03-13:
Hello again.

> _Can I only use the two joystick ports ? So I use the amiga keyboard and only convert mouse and joystick to usb. Is this possible ?_

This is possible, yes; the keyboard aspect is not mandatory, and as a result, you do not need to internally mount it if you just want the controller port access.

Joystick emulation hasn't been finished, and I have smatterings sitting in a git repository on my dev machine; I've been focussing on full-HID support for the mouse for a while and haven't yet finished that but it's closest to release.

What I'm saying is, mouse support is the only thing other than keyboard right now 😄 

> _Now anything is connected, but it does not work. Where should I insert the sleep function ?_

It's worth checking that the USB keyboard has been picked up and that there's not some sort of issue. If you have a 3.3V serial adapter (FTDI or similar), then pins 1 & 2 on the pico provide serial, and by default, unless you've edited [`CMakeLists.txt`,](https://github.com/borb/amigahid-pico/blob/main/CMakeLists.txt) it's turned on.

If you've built one of my PCBs, then the serial pins are marked on IDC connector J2.

Another good indicator is turning on caps lock; if the LED illuminates, it indicates that the firmware is talking to the keyboard, and caps lock is triggered via command in the event loop.

If you want to increase the startup pause, look in [`keyboard_serial_io.c`](https://github.com/borb/amigahid-pico/blob/main/src/platform/amiga/keyboard_serial_io.c). Line 96 has a 1s sleep (1000 ms). Bump it to, say, 3000ms and that should cover it.

Out of interest, have you customised anything in [`CMakeLists.txt`,](https://github.com/borb/amigahid-pico/blob/main/CMakeLists.txt)? Just wondering if the issue is something else.