borb/amigahid-pico #27: Support keyboard and mouse in parallel

evlog, 2023-08-25:
Great project. Can I connect in parallel both a keyboard and mouse on the usb type c, input using a hub?

borb, 2023-08-27:
Hello; I'm not sure which USB type C input you're referring to - are you referring to the PCB design where a type A socket sits onboard adjacent to the micro USB port on the Pico board?

However, yes, a hub is supported. I do note that I have not had success using USB3 hubs. My A1200 uses an amigahid-pico board for both keyboard and mouse concurrently. The eventual design is intended to provide joystick support via controller conversion without having to plug/unplug any devices, hence the OLED display for configuring which devices are routed to which ports. Furthermore, the mouse wheel codes need to be sent to the keyboard controller (two keycodes were reserved back in OS 3.1 for support at OS level without any additional software being installed). This will mean that using keyboard and mouse at the same time will offer scrollwheel support. The code is almost complete (on devel/mouse_hid_parser branch) and when I stop fiddling with my new A1200 board I will get it finished :laughing: 

I have an in-development board which I have a prototype for; it adds a new IC, three electrolytic capacitors, three ceramic capacitors and three resistors but breaks out to two type A ports and a 9-pin IDC header suitable for dual USB panel mount connectors, the sort found on PC motherboards. At present I have only two ports populated but it's working well. I'm looking to push this up to the repository shortly.

borb, 2023-08-27:
If you are happy with this answer and are not having any problems with using two devices concurrently, please feel free to close this issue. Otherwise I am happy to use it to discuss any issues you may be having.

evlog, 2023-08-27:
Thank you so much for your support. Yes I meant connecting a hub on the type A socket and connect both keyboard and mouse. Good to know that this is supported. Superb! Thanks again.