borb/amigahid-pico #31: Lantronics Spider KVM weird mouse issue

reinauer, 2024-03-04:
Hi,

I started using the amigahid-pico with a Lantronics Spider KVM. The 

keyboard works fine but the mouse does not:

- Mouse is not shown as detected on the LCD
- Mouse moves somewhat when I move the host mouse, but only in one direction (all the way down, or all the way to the right)
- If I keep the right mouse button pressed, the mouse moves almost perfectly <-- this is the weird part
- Mouse buttons seem to work fine

Also, is CTRL-AMIGA-AMIGA mapped to anything?

borb, 2024-03-04:
Hello,

Depending on which branch you're using, the mouse connection count is incorrect. I strongly suspect the `main` branch is a culprit at the moment - I have the same issue with the hid-pico in my A1200. Once I get a little free time I'll put the mouse motion counter update into the branch which contains full support for HID report mode mice and merge it, because that fixes it. I'll try to do it this week but family life means I'm not usually free until ~10pm most evenings.

I've seen the motion issue as and it's really weird. I've also seen the issue where holding the right mouse button down causes motion to smooth out. I'm going to need to rewrite the mouse motion output because it's junk and fairly stupid. I've hacked it a couple of times and it's just really ugly.

As for reset: ctrl-leftgui-rightgui or ctrl-leftgui-menu will trigger the reset line, regardless of which branch you're on. However, as Claude reported, I'm not asserting the minimum pulse length of 50ms so as a manual hack, hold the keys long enough to keep it held. If you're connecting to an A2000/A3000/A4000, reset is handled by asserting the clock line for a period of time and it's picked up by a small circuit and translated into a reset signal. That's a fairly simple change and also needs to be done.

reinauer, 2024-03-10:
Ha, I am moving slow on this project as well. Thank you for your pointers. I switched to the development branch and the mouse started working a loooot better! Thank you for the suggestion.

As for asserting the clock line, is that something that could be done unconditionally? Or would this require adding a compile time option for A2000/3000/4000? Or even better, can the amigahid detect the Amiga somehow?

I'll play around with with this on my  A3000.

reinauer, 2024-03-11:
This is probably not the right way to fix the issue, but I can confirm that the following patch will make CTRL-LeftWin-RightWin reboot the A3000

```diff --git a/src/platform/amiga/keyboard_serial_io.c b/src/platform/amiga/keyboard_serial_io.c
index 4088427..637b7ee 100644
--- a/src/platform/amiga/keyboard_serial_io.c
+++ b/src/platform/amiga/keyboard_serial_io.c
@@ -201,6 +201,10 @@ void amiga_assert_reset()
 {
     // ahprintf("[akb] *** RESET BEING ASSERTED ***\n");
     _keyboard_gpio_set(KBD_AMIGA_RST, LOW);
+    // Hold clock lo for 500us
+    _keyboard_gpio_set(KBD_AMIGA_CLK, LOW);
+    sleep_us(500000);
+    _keyboard_gpio_set(KBD_AMIGA_CLK, HIGH);
 }
 
 void amiga_release_reset()
```

borb, 2024-03-20:
That was the way I was intending to handle the reset line (I don't have a big-box Amiga to check with); I'll ask another user I know who uses this project to check it with their A2000.

I don't think there's really a point to making it configurable; at the very least, the .5s wait will help, since at the very least on my A1200 I frequently find that holding reset for too short a time will fail to reset the Amiga. Long term, once I've got the display menu+settings storage aspects working, I can make it a runtime configurable but certainly I don't believe there will be any ill effects using it on wedge Amigas.

I'll patch in the reset code now; it'll go into the development branch.

Should probably keep this issue open until development is merged down to main.

BennehBoy, 2025-10-25:
I've asked a friend with a big box amiga to scope the KBCLK behaviour on reset - whilst the delay above works for a reset it doesn't handle ctrl-amiga-amiga being held... this is important for wireless ROM switchers.  Hopefully it's an easy update to the code to accommodate (I'll submit a PR if so).

BennehBoy, 2025-10-27:
> It's a shame the MAME project doesn't have ROM dumps of the keyboard MCU that I could reverse engineer for behavioural cues (nowhere seems to, in fact) as that could answer a lot of questions as to how some slightly unusual features work.

LinuxJedi created a modern replacement for the 6570 last year, it's called ZonKEY... looks like he's implemented his updated logic with an AVR.. It doesn't look open source at the moment, but maybe he'd be willing to share some insights with you?

borb, 2025-10-30:
@reinauer wrote:
> I'm not claiming that the above patch was correct but it definitely does the issue of CTRL-AMIGA-AMIGA not working for me.
> 
> EDIT: This is on an A3000/A4000. I have not tested the device on an A500

I think the most important part of the patch was likely the minimum interval on holding the clock line; unfortunately I don't have a big box Amiga to test on so I can't confirm that side of things 😄 But from my perspective, I'm prepared to accept other people's patches if we can at least get confirmation that it works for at least one device with a wired keyboard and one with an internal keyboard. Your patch looked fine to me!

borb, 2025-10-30:
@BennehBoy wrote:
> PS scoping CLK on an A500 doesn't show any correlation to ctrl-amiga-amiga being pressed/held, it doe cycle low once the reset is underway though. A post online does seem to indicate that on the big box amigas a clk pulse initiates the reset hence why the workaround above works for a quick reset... I'm wondering if perhaps the desired outcome is to simply remove the sleep & add `+ _keyboard_gpio_set(KBD_AMIGA_CLK, HIGH);` to the amiga_release_reset() function.

AHH, the missing piece of the puzzle. That's spot on, I hadn't thought of that. I'd looked at the line setting rst back to high and conflated it with clk high. You're quite right. Both clk and rst are required.

> I suspect that the actual process of sending a keycode needs to be in a conditional that checks if in_reset is true, otherwise the code will try to send key presses for ctrl or lamiga or ramiga whilst we're in reset and change the clock state thus negating the held status signified by low clock. I'll experiment later.

No, I _think_ that still needs to be sent. I'm not sure if AmigaOS does anything in ROM to very quickly sync any buffers to disk.

I note that back in the era when I ran Linux on my A1200, the reset routine complained very loudly about how C-A-A was a bad idea under Linux because the time between noticing the three fingered salute (and thus triggering runlevel 6) and the reset line being asserted was vanishingly small, and thus the risk of filesystem corruption was too high; therefore, the MCU firmware _did_ send the C-A-A to the Amiga in spite of handling the reset line by itself. I'd like to maintain that compatibility. I'd also like to ensure that the pause between C-A-A and reset assert is implemented at some point, hopefully correctly.

BennehBoy, 2025-10-30:
> I note that back in the era when I ran Linux on my A1200, the reset routine complained very loudly about how C-A-A was a bad idea under Linux because the time between noticing the three fingered [...]

I suspect this might be chasing after functionality that isn't terribly well used in the Amiga as is.

However... this can be detected by checking the state of KB_DAT (I suspect independently of CLK), obviously in conjunction with sending the reset warnings.  See here -> https://amigadev.elowar.com/read/ADCD_2.1/Hardware_Manual_guide/node0178.html