borb/amigahid-pico #33: Strange mouse movement issues

PlunderBunny2000, 2024-12-31:
I'm using an old USB Microsoft IntelliMouse with my freshly-made amigahid-pico and it mostly works, but I'm running into a couple of strange issues: horizontal movement is a little jittery normally, and when I left-click I can't move the mouse horizontally AT ALL. Is this in any way a known thing, or do you have any idea what might cause this?

PlunderBunny2000, 2024-12-31:
Additional info: Can confirm that the mouse works just fine on other PCs, and can confirm that the same problem occurs when I use the mouse on another amigahid-pico board

borb, 2025-01-01:
Hello; this may be similar to the mouse motion issue mentioned in #31 - could you try building the development branch rather than main and see if the behaviour is fixed?

borb, 2025-01-01:
For reference: I have seen this behaviour with that code as well, including an unusual preference for the mouse motion to move in diagonals. This was owing to poorly nested conditionals in [quad_mouse.c](https://github.com/borb/amigahid-pico/blob). I'm running development on the board attached to my A1200 and it's a lot smoother.

borb, 2025-01-01:
Oh, and before I forget: Can you confirm which revision PCB you are using?

PlunderBunny2000, 2025-01-01:
I'm using PCB Rev. 5, and I've solved one of the problems myself, just by switching from the IntelliMouse to a different mouse. I was looking around other places to try to solve this problem and saw a chart that mentioned that the IntelliMouse doesn't quite comply 100% with the HID standards? 
So I thought maybe the mouse itself was responsible, switching to a new mouse fixed the horizontal jitteriness, although I still can't move the mouse horizontally when I have the left mouse button held down. That remaining problem definitely seems like it could be caused by poorly nested conditionals... I took a look at your main branch code but I am by no means a coder so I couldn't really tell if that's the problem, but I could understand if nesting of the conditionals might cause this behavior.
I'll try out the dev branch code and let you know if that clears things up both with the IntelliMouse and otherwise!

PlunderBunny2000, 2025-01-01:
Switching to the dev branch code fixed the last remaining problem, with having no horizontal movement while left-clicking. The movement does get a little weird when click-dragging, it feels like it doesn't want to move diagonally while click-dragging, but that's a more minor issue, everything is now at least functional with the dev branch code!

borb, 2025-01-02:
I have an IntelliMouse 5000 (the black laser variety) which I can give a try to reproduce the unusual motion; I'm not sure if it deviates from HID but for the most part, the version of amigahid-pico you're using uses the "hidbp" (boot protocol), which is a simplified architecture intended to be used by firmwares or operating systems prior to loading the "proper" HID driver.

devel/hid_mouse_parser is a branch currently being finished with a slight rewrite of the quadrature pulse mechanism but will culminate in moving to using HID reports rather than boot proto. It's been in progress for a long while now and I need the appropriate time and ability to clear enough desk space and power sockets nearby to get an Amiga running long enough to test it adequately :smile:  Soon, I really hope, because I'd like to move to controller support.

Keeping this issue open until development is merged into the main branch.