borb/amigahid-pico #36: Where does the D+ and D- go from the USB connector?

mamejay, 2025-08-12:
As the title states.  Checked the PCB and cannot see the data lines connecting to anything.

Am I missing something?

borb, 2025-08-12:
Hello @mamejay (and thank you @M1kerochip!)

This all depends on which D-/D+ you are referring to.

The small cutout beneat the Pico is intended to be filled so that it solders cleanly to the test pads on the rear of the Pico. Should look like this.

[image]

There is a position for a USB Type A header on the top side of the PCB, marked J4. The D-/D+ lines from the test pad run to this connector. End result should look like this:

[image]

As an alternative, you can fit 2.54mm pitched IDC header, four pin, and use a panel mount connector (the kind you get attached to PC motherboards for additional ports)

I'm sorry I didn't include a BOM in any of the PCB revisions. This was my first attempt at PCB design and is somewhat amateurish. I've been focussing on getting the MCU software improved with better support for mice and scroll wheels lately, so intend to revisit the PCB designs as well as producing something better suited to the Amiga 1200. At which point, I will include a BOM and better assembly instructions.

Please let me know if this answers your question.

mamejay, 2025-08-12:
Thanks so much!  I actually feel a little bit silly now.  I was trying to design a compact version of this to just do keyboard only, by using your kicad files and removing what I did no require.  Now I understand that the USB-A connection is just a duplicate of the onboard micro-usb of the PICO

M1kerochip, 2025-08-12:
https://github.com/wiretap-retro/A2000-USB-Keyboard-Adapter-RP2040

borb, 2025-08-13:
> I was actually looking for an external solution. I did check out wiretap's one but that uses a PiStamp which as not as readily available where I live and are a little pricey. I do like his solution though

All you require is clock, data and reset in that case; the level shifters will still be required but it should be simple enough.

I haven't had a change to revisit level shifting away from using discrete FETs and need to look in future into using an IC to provide sufficient signals, but I note that I did have trouble with using TXS0108 in that the voltage level on the reset line dropped too low and kept holding the Amiga in reset (or spontaneously resetting).

Do you mind if I ask which Amiga you are intending to attach this to?